nengo/nengo-fpga #2: Cleanup parameter npz files

bmorcos, 2018-08-08:
Parameters are transfered to the FPGA using an npz file. We should implement a system to cleanup old npz files so they don't buildup indefinitely.

xchoo, 2018-08-08:
I put in new exit handling code to deal with the `npz` files as part of the refactoring / rework of the individual `nengo-de1` and `nengo-pynq` repos (to handle the board side cleanup), and in the `nengo-fpga` repo to handle the host side cleanup [here](https://github.com/nengo/nengo-fpga/blob/2834e3c453e8924d45751ce376811a89c3d50785/nengo_fpga/simulator.py#L68). If this isn't working for you, then there's an issue with how the code is working, and we should find out how to reproduce it.

bmorcos, 2018-08-08:
Awesome. Looks like I just didn't have the latest commit locally.

closed 2018-08-08 by bmorcos