nengo/nengo-fpga #3: Do we need to have a password field in the config?

bmorcos, 2018-09-18:
The `fpga_config` file has the password for the device written out in plain text. While this *probably* isn't a big deal in most cases (since it's the default user/pw on a local network) we might want to look into other ways of authentication?

Just putting this thought here so it isn't lost.

xchoo, 2018-09-18:
Hmm. Yeah, we could get the user to set up an ssh-key from the host PC, so that passwords are not necessary at all.

bmorcos, 2018-12-17:
As mentioned by @xchoo, ssh removes password requirement but now we must decide on default behaviour (always use ssh? no need for password at all?)

xchoo, 2018-12-17:
Just as a point of clarification, it always uses SSH, it's whether or not to use the SSH keys for the SSH authentication process.