nengo/nengo-fpga #54: Not able to connect nor ping the FPGA on first boot.

pedrombmachado, 2020-04-10:
I followed the installation steps for both DE1 images (i.e. 20190521_DE1_0.2.0b.img and 20200124_DE1_v0.2.1.img) and I cannot ssh into the FPGA. In fact, I cannot even log into the board via the serial port (i.e. ttyUSB0 because it never gets into the login stage).
```
[  OK  ] Started System Logging Service.
[  OK  ] Started Kernel Logging Service.
[  OK  ] Started Timestamping service.
         Starting Network Time Service (one-shot ntpdate mode)...
         Starting Login Service...
[  OK  ] Started D-Bus System Message Bus.
         Starting Network Service...
         Starting Connection service...
         Starting (null)...
         Starting Avahi mDNS/DNS-SD Stack...
[  OK  ] Started Network Time Service (one-shot ntpdate mode).
[  OK  ] Started Network Service.

```

xchoo, 2020-04-11:
The latest SD image for the DE1 platform is 20200124_DE1_v0.2.1.img, so you should be using that.

As for why it is not going into the login screen, check that the dipswitches have been set correctly, according to: https://www.nengo.ai/nengo-de1/getting-started.html#setup-the-de1-soc-board

Also, just to check, do you see 4 red leds light up and follow a binary counter pattern after you power cycle the board?

pedrombmachado, 2020-04-11:
HI @xchoo ,
  It worked, the problem were the dipswitches. Many thanks

closed 2020-04-11 by pedrombmachado