nengo/nengo-fpga #74: Please add support for DE0-nano-CycloneV SOC board #enhancement

shumtravi, 2021-03-01:
Please add support for [DE0 Nano CycloneV SOC](https://www.terasic.com.tw/cgi-bin/page/archive.pl?Language=English&CategoryNo=167&No=941), it's cheap and popular.

[image]

Spec:

> FPGA Device
> Altera Cyclone® V SE 5CSEMA4U23C6N device
> Serial configuration device – EPCS
> USB-Blaster II onboard for programming; JTAG Mode
> 2 push-buttons
> 4 slide switches
> 8 green user LEDs
> Three 50MHz clock sources from the clock generator
> Two 40-pin expansion header
> One Arduino expansion header (Uno R3 compatibility), can connection with Arduino shields.
> One 10-pin Analog input expansion header. (shared with Arduino Analog input)
> A/D converter, 4-pin SPI interface with FPGA
> HPS (Hard Processor System)
> 925MHz Dual-core ARM Cortex-A9 processor
> 1GB DDR3 SDRAM (32-bit data bus)
> 1 Gigabit Ethernet PHY with RJ45 connector
> USB OTG Port, USB Micro-AB connector
> Micro SD card socket
> Accelerometer (I2C interface + interrupt)
> UART to USB, USB Mini-B connector
> Warm reset button and cold reset button
> One user button and one user LED
> LTC 2x7 expansion header

bmorcos, 2021-03-03:
Hi shumtravi and thank you for your interest!

We have experimented with the DE0-nano but it does not currently integrate well into the software stack we use, so it would involve more development effort to support. In addition, the smaller size (memory, LUts) would further limit the possible ensemble size placed on the device. So unfortunately, this is not part of our current plan.

Closing since this is already on our radar, but not something we are currently planning to support.

closed 2021-03-03 by bmorcos

shumtravi, 2021-03-06:
Hi bmorcos! Thank you for clarification.