nengo/nengo-fpga #76: Support for Ultra-Low Power FPGAs such as the Lattice iCE40UP5K

contrebande-labs, 2021-04-23:
Hello, I just learned about the Nengo project trying to figure out a way to build something like the [Loihi-based VOC sensor artificial nose from Imam & Cleland (2020)](https://www.nature.com/articles/s42256-020-0159-4).

I was wondering if it was possible to port the `nengo-fpga` backend to the Lattice iCE40UP5K, or even better an array of such devices?

Thanks !

xchoo, 2021-04-26:
Hi! The NengoFPGA "backend" consists of two components that work together. The first component (`nengo-fpga`) is just an interface between Nengo and any backend that supports the API. The API itself can be found [here](https://github.com/nengo/nengo-fpga/blob/master/nengo_fpga/networks/fpga_pes_ensemble_network.py#L681), and is basically just a UDP socket that sends and receives floating point data from the actual device backend. Adapting the NengoFPGA interface to support another device shouldn't require any work at all, since the API needs to be implemented by the device backend.

The second component is the device backend, of which there are currently two: [NengoDE1](https://github.com/nengo/nengo-de1) and [NengoPYNQ](https://github.com/nengo/nengo-pynq). Supporting another device, such as the Lattice iCE40UP5K would require development of an equivalent backend for the Lattice device. Unfortunately, development for NengoFPGA and NengoFPGA related repositories has been temporarily stopped in order to prioritize other Nengo projects. However, if you would like to develop the Lattice backend yourself, integrating it into the NengoFPGA framework shouldn't be too difficult.

closed 2021-04-26 by xchoo

contrebande-labs, 2021-04-26:
Hi, I want get as close as possible to what the authors of the paper cited above are doing with the Loihi chip with a cluster of sensor boards with the Lattice iCE40UP5K FPGA on them instead of the Loihi chip. It seems the links to the `nengo-de1`and `pengo-pynq` you referred to give a 404 error now. Their documentation is still available. Where do I start if I want to make a NengoFPGA backend for the Lattice iCE40UP5K? Do the DE1 and PYNQ projects contain some HDL source I could get some inspiration from? If so, could this source be made publicly available? Thanks.

xchoo, 2021-04-26:
Oh. Haha. Apologies. The NengoPYNQ and NengoDE1 repositories are private (because they contain the interfaces to the devices themselves), so I've modified the links to point to the public facing documentation. 

> Do the DE1 and PYNQ projects contain some HDL source I could get some inspiration from? If so, could this source be made publicly available?
Yes, the DE1 and PYNQ projects do contain the HDL source code. Unfortunately, we don't have any plans to make those publicly available at this current time.

> Where do I start if I want to make a NengoFPGA backend for the Lattice iCE40UP5K?
If you want to develop a backend for the Lattice device, I'd recommend familiarizing yourself with the [NEF algorithm](https://www.nengo.ai/nengo/examples/advanced/nef-algorithm.html). We use this to compress vector information to and from the FPGA device. In the NengoPYNQ and NengoDE1 implementations, we are essentially simulating 1 Nengo ensemble on the FPGA device.

contrebande-labs, 2021-04-26:
Got it, thanks. And can you tell me where is the best place to discuss NEF ?

xchoo, 2021-04-26:
You can post questions on our forums! https://forum.nengo.ai/
(Although I see that you may have already made a post) 😄

contrebande-labs, 2021-04-27:
Ah yes. No it's not me. But it's a nice coincidence. Thanks!